Cartographer3D/cartographer-klipper #10: I had an issue with idm.py due to (f" syntax in line 892 and 1997

PopoTheBest, 2024-02-06:
Had to remove in line 892 and 1997 the "f" in the lines starting with "(f"" or I had an error in Klipper.
Is it normal ?

Thanks
Philippe

RichardTHF, 2024-02-19:
You are likely running Python2, it requires Python3.

Tomblarom, 2024-03-02:
I had a similar issue and used the following commands to upgrade from python2 to python3. The prerequisite for this is, that python3 is already installed.

```
sudo service klipper stop

# remove current klippy-env environment
cd ~
rm -rf klippy-env

# create new venv
virtualenv -p python3 klippy-env

#install new Klipper venv
cd ~/klippy-env
bin/pip install -r ../klipper/scripts/klippy-requirements.txt
```
Copied from https://klipper.discourse.group/t/process-for-migrating-to-python3/5292/3

RichardTHF, 2024-03-03:
Thank you @Tomblarom, I will add to the docs shortly.

Tomblarom, 2024-03-03:
> Thank you @Tomblarom, I will add to the docs shortly.

On my raspbian (Debian 10 Buster) was python3.7 installed and later I had trouble installing numpy. Reinstalliung klipper also failed and I got issues like these:

```
ModuleNotFoundError: No module named 'pkg_resources'
or DeprecationWarning: pkg_resources is deprecated as an API
or DistributionNotFound: The 'virtualenv==15.0.3' distribution was not found and is required by the application
```

All of this was an indicator that my system was too old and I ended up upgrading to Debian 11 Bullseye, which comes with python3.9. I might even lift it to Debian 12 Bookworm.

RichardTHF, 2024-03-03:
I know that there has been some reported errors with Klipper and Bookworm, so I can't 100% say that it's a good upgrade. 

I have been using MainsailOS, and have loved it.

closed 2024-03-17 by RichardTHF