Cartographer3D/cartographer-klipper #13: CARTOGRAPHER_CALIBRATE applying calibration to default model

patrezp, 2024-04-18:
I think that the current behavior of the CARTOGRAPHER_CALIBRATE is not something that is 'user friendly'
I would expect to see the current loaded model to be calibrated and not the default one.

Do you think that this is something that can be easily changed? I've burned myself couple of times

RichardTHF, 2024-04-29:
Sorry for the slow response to this, I will look into that. Thank you for the suggestion.

andremalenfant, 2024-08-30:
I just realized the same and it was confusing me why I would get nozzle scratching my surface on hot prints. I just figured out the same and checked the code. It indeed ignores the current active model even though the documentation clearly states that the calibration has to be done on all models. I am not sure how can be done other than copy pasting the data from the save confirmation in klipper and overriding the model manually.

closed 2024-11-16 by krautech