Cartographer3D/cartographer-klipper #14: Cartographer sensor not receiving data or Communication timeout during homing probe

devolf92, 2024-04-22:
Receiving this error after bed mesh calibrating or sometimes random even while sensor is not in use. Cant google anythig about this error.
Firmware is stock CAN, connected to ebb sb2209 can.
Klipper is uptodate

sometimes i got Communication timeout during homing probe while QGL on my voron 2.4
looks like things getting worse wen enclosure get hotter

[image]

cruiten, 2024-05-30:
I recommend that you open a support issue on the Cartographer Discord Help and Support channel: https://discord.gg/HBmJnk4y
(This link will only be good for seven days)

You can also check out this excellent guide: https://canbus.esoterical.online/troubleshooting/timeout_during_homing_probing.html

devolf92, 2024-05-31:
Thanks for your reply
I'm currently using fix with TRSYNC_TIMEOUT since this issue started. But this dirty status driving me mad.
Looks like this is fresh new guide or i miss it, thanks!

closed 2024-11-16 by krautech