Cartographer3D/cartographer-klipper #29: BED_MESH_CALIBRATE METHOD=scan crashes latest klipper with `'CartographerProbeWrapper' object has no attribute 'get_status'`

iMartyn, 2024-08-22:
New install, as of today, cannot help pinpoint what change broke this because I have never had a carto before.

cartographer-klipper version : v1.0.0-50-gf9c27d38 
klipper version : v0.12.0-286-g81de9a86 
[image]

iMartyn, 2024-08-22:
So for reference if anyone comes to this issue... don't add a `METHOD=` on cartographer, it's unneeded and crashes.

krautech, 2024-08-22:
@iMartyn I'll or someone else will look into making sure no extra commands cause this to error out and instead get ignored.

👍🏻

iMartyn, 2024-08-22:
Yes, so I was looking for recommended settings to put in `[bed_mesh]` because I was moving from klicky to cartographer and my thought process was "well, the point of a bed scanner like cartographer is to have more points, so what should I change, and the FAQ for no bed mesh sent me to [the klipper docs](https://www.klipper3d.org/Bed_Mesh.html) and the [section on bed scanners](https://www.klipper3d.org/Bed_Mesh.html#surface-scans) has the detail on adding `METHOD=`.  Perhaps an example `[bed_mesh]` section on the cartographer docs site might avoid people getting docs from klipper where things are sometimes not accurate for the specifics?

krautech, 2024-11-16:
Sorry i'm replying to this so late. [bed_mesh] examples have been added to docs :)

closed 2024-11-16 by krautech