Cartographer3D/cartographer-klipper #36: [FR] 2 small adjustemnts to CARTO_TOUCH output

TomW1605, 2024-09-08:
these are just 2 samll things i have noticed that i think would be nice to have improvements

1) when running CARTO_TOUCH show what retry it is up to. it shows what tap within each retry but not what retry it is.

2) either show the deviation to the number of decimal places it is being compared in or round it to the number of decimal places it is shown it before compareing it. not a big one but seeing `// Deviation of 0.0100 exceeds tolerance of 0.0100, retrying...` is very frustrating. im assuming the actual deviation is something like 0.01005... and it is just not showing the full precision. it is also possible it is being compared with a < rather than <= and if that is the case i think it would also be worth changing.

krautech, 2024-09-08:
I'll come up with something.
The idea behind not showing more than a couple decimal places is because at those fine levels it doesn't matter and people start number chasing.

I'll add what retry it's up too, that's a good idea and I dunno why I didn't already do it since it's frustrated me a few times 🤣

TomW1605, 2024-09-08:
yeah, i dont really like the idea of showing a lot of decimals but seeing 0.01 exceeds 0.01 is pretty frustrating

krautech, 2024-09-16:
[Added to KrauTech Nightly Branch](https://github.com/Cartographer3D/cartographer-klipper/commit/d503b4743de55d5401157cfb9997692f774bc6c2)

krautech, 2024-11-15:
PR #48 & PR #45

closed 2024-11-15 by krautech