Cartographer3D/cartographer-klipper #37: Disabling cartographer mcu during print

viktor-zhuromskyy, 2024-09-29:
My actual issue is that the Cartorapher probe device is constantly loaded between 35-45% and adding to the heat collected off the heated bed and printed part.

I do realize the cartographer probe is needed only when probing and meshing, and during the print there is no actual use of it.

Is there any way to suspend / disable the probe during actual print, lets say starting with second layer or so? Is it feasible?

[image]
[image]

closed 2024-11-16 by krautech

Maokaman1, 2024-11-16:
Creality K1 FW variant `Survey_Cartographer_K1_USB_8kib_offset.bin` is still v5.0.0. Is this intended?

krautech, 2024-11-17:
I'll make sure 5.0.2 is added today for K series.

But also, let me just check if the fix for K series isn't just in the normal firmware now