Cartographer3D/cartographer-klipper #39: mcu 'scanner': Unable to extract params from: config_lis2dw

Travis90x, 2024-11-14:
My printer.cfg
IDM scanner flashed with 
~/cartographer-klipper/firmware/v2-v3/survey/Survey_Cartographer_USB_8kib_offset.bin

my bootloader serial 
/dev/serial/by-id/usb-katapult_stm32f042x6_2C0013001743565537353020-if00

[scanner]
serial: /dev/serial/by-id/usb-Cartographer_614e_2C0013001743565537353020-if0
x_offset: 0     
y_offset: 15   
backlash_comp: 0.5
calibration_method: touch
sensor: cartographer
sensor_alt: carto
scanner_touch_z_offset: 0.05  
mesh_runs: 2

[lis2dw]
cs_pin: scanner:PA3
spi_bus: spi1

[resonance_tester]
accel_chip: lis2dw
probe_points: 150, 150, 150 
min_freq: 25
max_freq: 150

[bed_mesh]
zero_reference_position: 175, 175

[image]

if I disable 
[resonance_tester] and [lis2dw]

[image]

RichardTHF, 2024-11-14:
fixed with latest firmware update :)

closed 2024-11-14 by RichardTHF

Travis90x, 2024-11-15:
> fixed with latest firmware update :)

Thanks for the fix 
[image]