Cartographer3D/cartographer-klipper #61: bug: 5.0.2 touch not working as described

krautech, 2024-11-17:
**DESCRIPTION**

`CARTOGRAPHER_THRESHOLD_SCAN` would move down, then jitter, repeating. I believe it was false positive repeating itself causing every threshold to fail.

The default touch speed of threshold is 3.

`CARTOGRAPHER_TOUCH` would also fail, default touch speed 3.
- When this is set to SPEED=2 it would pass.

yell3D, 2024-11-17:
did 2x auto model 2x z offset, no issues using default treshhold (2500?)

* Trident
* Mjönir
* Chube
* Creality small Z stepper + coupler
* textured Pei + regular Voron bed

# Versions
[image]
[image]

# Touches
[image]
[image]
[image]
[image]

# Printer.cfg

```python
## cartographer.cfg
[scanner]
canbus_uuid:         replaceme
sensor:           cartographer
sensor_alt:             beacon     # > you can have sensor_alt: beacon if you really wanted to shit stir 😄
calibration_method:      touch
default_model_name:    default
mesh_main_direction:         y
x_offset:                    0
## Yell Customs
samples_tolerance_retries:  20
samples_tolerance:           0.02
samples_retract_dist:        3.0
samples:                     6
scanner_touch_retract_speed: 5     # stock values not equal to threshhold scan lift speed
scanner_touch_tolerance:     0.009
scanner_touch_max_retries:  10
scanner_touch_max_temp:    155
scanner_touch_move_speed:  150

[temperature_sensor Cartographer_MCU]
sensor_type:   temperature_mcu
sensor_mcu:            scanner
min_temp:                    0
max_temp:                  105

[adxl345]
cs_pin:            scanner:PA3
spi_bus:                  spi1
axes_map:             x, -y, z

# override in printer.cfg after include if high temp
#[adxl345]
#cs_pin:            scanner:PA0
#spi_bus:                  int2

[stepper_z]
homing_retract_dist:         0
endstop_pin: probe:z_virtual_endstop

[bed_mesh]
speed:                     400
probe_count:                51,  51
#zero_reference_position:

## printer.cfg
[include ./addons/cartographer.cfg]
[scanner]
canbus_uuid:            6e1aedba6383
y_offset:              27.8
backlash_comp:          0.00472
```

krautech, 2024-11-17:
Thanks @yell3D it seems like its only affecting some people. Want to investigate further before putting it out. It seems like the i2c changes mightve changed how frequency is being affected by certain things.

My threshold doubled using it while others halved. We have a solution in the works.

closed 2024-11-20 by krautech