Cartographer3D/cartographer-klipper #77: Reduce cartographer polling when not in use

pellcorp, 2024-11-19:
For the Creality K1 series especially the Cartographer stresses the weak MCU even during printing when the carto is not even in use, it would be ideal if the cartographer could limit polling for temp and the like to an absolute minimum when not being used for homing or meshing.

krautech, 2024-11-20:
I started work on this but had to delay due to 5.0.2/lis2dw breaking. Its something we're trying to figure out. I'll keep this updated with progress. All firmware related changes happen at [Firmware Repo](https://github.com/Cartographer3D/MCU-Firmware---Based-on-Klipper/)

This hopefully will be something implemented into v1.1.0 however it might require long term testing on the beta channels.

RichardTHF, 2024-11-20:
I have spent a LOT of time trying to work on this, and have never got it working properly. 

5.0.2 might be better for K1, but will remain to be seen.

krautech, 2024-11-20:
Yeah I got as far as EVERYTHING working, except calibration. Homing, Touching, Bed Mesh I had working. Calibration didn't like it.

I'll be working on this more tomorrow

pellcorp, 2024-11-20:
A bunch of Creality K1/K1C/K1Se and K1M will forever be in your dept :-)

pellcorp, 2024-11-21:
> Yeah I got as far as EVERYTHING working, except calibration. Homing, Touching, Bed Mesh I had working. Calibration didn't like it.
> 
> I'll be working on this more tomorrow

and this would include start print cartographer_touch too I assume?

krautech, 2024-11-21:
Yeah the start print macro is just the same call.

I haven't done any work today, needed a break, was falling asleep on the PC.

Spent like 40+ hours on carto code this week already 😂

krautech, 2024-11-26:
@RichardTHF I'll need 5.1.0 pushed to the MCU firmware github so I can play around with this on the 5.1.0 firmware.

@pellcorp this will probably come in V1.2.0 as I'd rather devote a bit more time with R&D for it.