Cartographer3D/cartographer-klipper #141: Support danger klipper is_non_critical

pellcorp, 2024-12-03:
It is possible to set is_non_critical flag on `[scanner]` config to prevent klipper stopping if the cartographer disconnects, but even though danger klipper does allow the cartographer to reconnect, klipper will crash when you try to next home Z.   So the scanner.py needs to properly support reconnect mode of danger klipper.

krautech, 2024-12-04:
https://github.com/Cartographer3D/cartographer-klipper/blob/master/scanner.py#L479

Is where it crashes which is handled by the klipper side of things
I had it successfully reconnect and displaying data however when it would call this function from klippy it would force klipper restart without an error log or error message.

Maybe someone else could investigate. I had it down to it being a klippy side issue rather than scanner.

pellcorp, 2024-12-05:
So when trying to home Z on the printer after the cartographer reconnects we actually get a proper crash with a page fault

[image]

krautech, 2024-12-06:
> So when trying to home Z on the printer after the cartographer reconnects we actually get a proper crash with a page fault [...]

interesting, I dont know what we can do with this information however, looks like a kernel issue? I'll do some digging

krautech, 2024-12-06:
what FW and Cartographer-Klipper code is this with?

pellcorp, 2024-12-09:
I'm closing this issue, its no longer relevant for Simple AF

closed 2024-12-09 by pellcorp