Cartographer3D/cartographer-klipper #144: Delta printers probe error

Sab-tech-lab, 2024-12-05:
While working on getting Cartographer to work on my v400 I noticed it was missing the probe declaration.
Adding this line:
```
def get_status(self, eventtime=None):
        return {“name”: “probe”}
```
to ScannerWrapper class of the scanner.py file.

I was able to get the probe working, could you please add this small change in the next version of the software?

krautech, 2024-12-05:
@Jomik

Sab-tech-lab, 2024-12-06:
There's probably also a problem with the automatic z-offset calculation, but I haven't been able to figure out where the problem is yet; I suspect the problem is related to the delta kinematics and its way home that involves bringing the printer head to the maximum possible height instead of down near the print bed. I am trying to analyze what is happening in the offset calculation that seems too low.

krautech, 2024-12-06:
I don't think carto or bed scanners are generally working on delta printers as a whole are they?

It's impossible to scan the bed correctly

Sab-tech-lab, 2024-12-06:
> I don't think carto or bed scanners are generally working on delta printers as a whole are they?
> 
> It's impossible to scan the bed correctly

Sorry but why do you say that? My configuration works correctly, I just have problems with the automatic z-offset calculation.
[image]

Bed mesh created with cartographer:
[image]

Sab-tech-lab, 2024-12-06:
> Seems like two separate issues here?

Actually you are right, shall I open another issues?

krautech, 2024-12-06:
> > Seems like two separate issues here?
> 
> Actually you are right, shall I open another issues?

Yeah this one is a PR ready for merge. Can you make another `issue` for other problem please?

krautech, 2024-12-06:
Resolved with #156

PR #156 (2024-12-06): Implement probe get_status
## Description

Fixes #144