Cartographer3D/cartographer-klipper #149: CARTOGRAPHER_THRESHOLD_SCAN seems to be finishing too early with No suitable threshold was found during the scan.

pellcorp, 2024-12-06:
Running the following:

```
CARTOGRAPHER_THRESHOLD_SCAN MIN=1500 MAX=5000
```

We encounter this result:

```
$ CARTOGRAPHER_THRESHOLD_SCAN MIN=1500 MAX=5000
18:32:33 
// Testing Threshold value 1500...
18:32:41 
// Attempt 1/5 failed with deviation 0.1039.
18:32:47 
// Attempt 2/5 failed with deviation 0.0184.
18:32:52 
// Attempt 3/5 failed with deviation 0.3285.
18:32:57 
// Attempt 4/5 failed with deviation 0.3273.
18:33:03 
// Attempt 5/5 failed with deviation 0.1996.
18:33:05 
// Testing Threshold value 1750...
18:33:08 
// Attempt 1/5 failed with deviation 0.4813.
18:33:13 
// Attempt 2/5 failed with deviation 1.0953.
18:33:19 
// Attempt 3/5 failed with deviation 0.0562.
18:33:24 
// Attempt 4/5 failed with deviation 0.4152.
18:33:29 
// Attempt 5/5 failed with deviation 0.6598.
18:33:31 
// Testing Threshold value 2000...
18:33:34 
// No suitable threshold was found during the scan.
```

I will ask the Simple AF user to upload logs soon

Brakisshuro, 2024-12-06:
Here is requested logs and cfg's
[klippy.zip](https://github.com/user-attachments/files/18033215/klippy.zip)

Brakisshuro, 2024-12-06:
pictures
[image]
[image]

krautech, 2024-12-06:
Did speed=2 fix this?

pellcorp, 2024-12-06:
Not sure, have asked Brakiss to run, will post here if it makes a difference

pellcorp, 2024-12-06:
Seems like a bug that it would bail out before MAX value though

krautech, 2024-12-06:
It does. I'm looking at the code now to spot it.

Can you get it ran at DEBUG=1 as well? And then a copy and paste to a txt of the output would be amazing

pellcorp, 2024-12-06:
will do, might not happen till later on today, I think Brakiss is from the US

Brakisshuro, 2024-12-06:
Good morning, I am in the US. How do I run debug?

krautech, 2024-12-06:
@Brakisshuro just add `DEBUG=1` to the end of you `CARTOGRAPHER_THRESHOLD_SCAN` command

Brakisshuro, 2024-12-06:
Ok will upload shortly

Thank you,

Shane Greenfield - REALTOR®
Living CDA Realty Group - eXp Realty

208.640.1269

-Sent from my mobile device, so please excuse any tyP0'$

krautech, 2024-12-06:
Yeah okay that doesnt make sense does it ^.^

Brakisshuro, 2024-12-06:
Here you go, failed almost immediately again
[Uploading klippy.log (1).txt…]()

krautech, 2024-12-06:
Did it finish uploading? It says its uploading still so i cant download it. I _THINK_ i know what it most likely is in the code thats wrong though

Brakisshuro, 2024-12-06:
Edited previous post

krautech, 2024-12-06:
I thought I did, but looking at your logs.. its happening at completely random and different points where it just doesnt seem possible. Like if it was the code and it was erroring at a certain point, then it would always be there. However in your case, it errors in different places.

```
                try:
                    probe_position = self.phoming.probing_move(
                        self.mcu_probe, homing_position, speed
                    )
                 **ERROR HERES #1**
                except self.printer.command_error as e:
                    if self.printer.is_shutdown():
                        self.trigger_method = 0
                        raise self.printer.command_error(
                            "Touch procedure interrupted due to printer shutdown"
                        ) from e
                    raise
                finally:
                    self.set_accel(max_accel)

                retract_position = self.toolhead.get_position()[:]
                retract_position[2] = min(retract_position[2] + retract_dist, z_max)
                self.toolhead.move(retract_position, retract_speed)
                self.toolhead.dwell(1.0)
                **ERRORS HERE #2**
                samples.append(probe_position[2])
                self.log_debug_info(
                    verbose,
                    gcmd,
                    f"Touch {len(samples)} result: {probe_position[2]:.4f}",
                )
```

krautech, 2024-12-06:
Now where #1 errors, it doesnt trigger the except block, nor does it reach finally.

and then #2 errors further down the code before it displays the touch result.

Brakisshuro, 2024-12-06:
Interesting, so do you want me to try that code?

krautech, 2024-12-06:
Need to investigate further, I dont know if its code related or not yet.

What shows up if you do a `PROBE_ACCURACY`

Brakisshuro, 2024-12-06:
10:47:32 
$ G28
10:47:38 
$ PROBE_ACCURACY
10:48:01 
// PROBE_ACCURACY at X:152.000 Y:145.000 Z:10.000 (samples=10 sample_retract_dist=5.000 speed=5.0 lift_speed=5.0 skip=0)
10:48:05 
// probe at 152.000,145.000,1.967 is z=1.972414
10:48:09 
// probe at 152.000,145.000,1.964 is z=1.964268
10:48:12 
// probe at 152.000,145.000,1.967 is z=1.967415
10:48:16 
// probe at 152.000,145.000,1.970 is z=1.970549
10:48:19 
// probe at 152.000,145.000,1.979 is z=1.979119
10:48:23 
// probe at 152.000,145.000,1.938 is z=1.940303
10:48:26 
// probe at 152.000,145.000,1.972 is z=1.973758
10:48:30 
// probe at 152.000,145.000,1.963 is z=1.963645
10:48:33 
// probe at 152.000,145.000,1.973 is z=1.973665
10:48:37 
// probe at 152.000,145.000,1.980 is z=1.979655
10:48:37 
// probe accuracy results: maximum 2.000826, minimum 1.994786, range 0.006040, average 1.998877, median 1.999390, standard deviation 0.001589

krautech, 2024-12-06:
Do you know how to switch git branches? Ive added some more debugging output to threshold scan but its on a different branch youd have to pull

Brakisshuro, 2024-12-06:
Mostly, I just switched my branches last night on Pells beta

krautech, 2024-12-06:
```
cd /usr/data/cartographer-klipper
git fetch
git switch fix/patch
cd ~
/usr/share/klippy-env/bin/python3 -m compileall /usr/data/klipper/klippy
/etc/init.d/S56moonraker_service restart
/etc/init.d/S55klipper_service restart
```

Brakisshuro, 2024-12-06:
ok do I need to update?

krautech, 2024-12-06:
```
cd /usr/data/cartographer-klipper
git fetch
git switch fix/scan-patch
cd ~
/usr/share/klippy-env/bin/python3 -m compileall /usr/data/klipper/klippy
/etc/init.d/S56moonraker_service restart
/etc/init.d/S55klipper_service restart
```

Sorry do this one, you shouldnt need to update or anything once you do. just try the scan again with DEBUG=1

Brakisshuro, 2024-12-06:
same error, but it was not even getting close to the nozzle.
[klippy.log](https://github.com/user-attachments/files/18042744/klippy.log)

krautech, 2024-12-06:
Doesnt look like debug worked again

`CARTOGRAPHER_THRESHOLD_SCAN DEBUG=1`

Brakisshuro, 2024-12-06:
Here it is
[klippy.log](https://github.com/user-attachments/files/18043070/klippy.log)

krautech, 2024-12-06:
Thanks, still doesn't seem to trigger anything.

Did your threshold scan work pre-beta? (5.0.0)

Brakisshuro, 2024-12-06:
it did, with the exception it failed half the time when I started prints

pellcorp, 2024-12-07:
The fix scan patch won't exist on my fork, Brakiss did you change the repo origin to cartographer.   Maybe show a screenshot of when you switch to make sure switching branches is actually working

krautech, 2024-12-07:
Oh does it pull from a fork? Derp 😂 

Okay this is hard to communicate here,  DM me on discord?

Tried to reproduce on my 3 systems over 6 hours and couldn't get it to do it at all.

krautech, 2024-12-07:
> The fix scan patch won't exist on my fork, Brakiss did you change the repo origin to cartographer.   Maybe show a screenshot of when you switch to make sure switching branches is actually working 

Completely my fault I should've doubled checked this

pellcorp, 2024-12-07:
```
cd /usr/data/cartographer-klipper
git remote set-url origin https://github.com/Cartographer3D/cartographer-klipper.git
git fetch origin 
git switch fix/scan-patch
cd ~
/usr/share/klippy-env/bin/python3 -m compileall /usr/data/klipper/klippy
/etc/init.d/S56moonraker_service restart
/etc/init.d/S55klipper_service restart
```

pellcorp, 2024-12-07:
Seems that branch the scanner.py has python syntax issue with indentation

krautech, 2024-12-08:
Should be up to date and fixed now

Brakisshuro, 2024-12-08:
So should I run a debug with your repository and then switch back over to the beta?

alihakkidemiral, 2025-01-06:
hi i try install it. is it right ?
[image]

alihakkidemiral, 2025-01-06:
and my printer head is broken, the nozzle has drawn a few pictures. the side of the eva3 that holds the rapido is shaking. probably because of this. i will fix it as soon as possible and reject it again. super glue time :D

alihakkidemiral, 2025-01-20:
> and my printer head is broken, the nozzle has drawn a few pictures. the side of the eva3 that holds the rapido is shaking. probably because of this. i will fix it as soon as possible and reject it again. super glue time :D

Yeap, my cartographer works perfect now. My printer head was broken and hotend was swinging. some time done at first time.

Kaperkapeike, 2025-04-07:
This should be fixed here, right?
I am currently experiencing exactly this behavior, the threshold scan sometimes aborts after one attempt