Cartographer3D/cartographer-klipper #157: Automatic z-offset not work on delta printer

Sab-tech-lab, 2024-12-06:
Printer: FLsun V400 with official Klipper builds.

I don't know if this can really be called a report, if not, close it instantly.
Maybe I misunderstood but on the official website it talks about:
" Survey Touch Now Available
Survey Touch is now currently available to all V2 & V3 Cartographer Probes*. Survey Touch is our new contact probe solution giving you Auto-Z offset, allowing you to quickly and easily switch between build plates and nozzles."

And into official documentation on page [Doc page command](https://docs.cartographer3d.com/cartographer-probe/survey-touch/settings-and-commands) i can read this:
"Z-Offset Calibration: Adjusts the Z-axis offset to ensure the nozzle is at the correct distance from the print bed for optimal printing." into CARTOGRAPHER_TOUCH command explanation section.

But then I don't understand why I should set the z-offset as given on this page: 
[Doc page](https://docs.cartographer3d.com/cartographer-probe/installation-and-setup/installation/first-print)

In any case when I use the "CARTOGRAPHER_TOUCH DEBUG=1" command anyway I do not get a correct z-offset but the nozzle always tends to scratch on the bed.

Did I misinterpret the documentation? 
In any case, I attach the output of the CARTOGRAPHER_TOUCH command (with the order of the lines reversed for more comfortable reading).

```
16:19 MOVEMENT_SPEED: 50
16:19 ACCEL: 100
16:19 RETRACT: 2
16:19 RETRACT_SPEED: 10
16:19 SAMPLES: 3
16:19 TOLERANCE: 0.01
16:19 RETRIES: 10
16:19 TOUCH_LOCATION_X: 0.0
16:19 TOUCH_LOCATION_Y: 0.0
16:19 THRESHOLD: 5000
16:19 Z_OFFSET: 0.05
16:19 MOVE: 0
16:19 Trigger Method: 1
16:19 Initial Pos: [0.0, 0.0, 10.0, 0.0]
      Homing Pos: [0.0, 0.0, 10.0, 0.0]
      Z MIN: -25.0
      Z MAX: 416.599
16:19 new Initial Pos [Initial Z - Z Max]: [0.0, 0.0, 416.599, 0.0]
      new Homing Pos [Homing Pos - Z Min]: [0.0, 0.0, -25.0, 0.0]
16:19 Current Accel: 10000
16:19 Initiating Touch Procedure...
16:19 Set Acceleration to: 100
16:19 Executing Touch 1 of 3 [0/10]
16:19 Touch 1 result: 406.0365
16:19 Reset Acceleration to: 10000
16:19 Deviation: 0.0000
      New Average: 406.0365
      Tolerance: 0.0100
16:19 Set Acceleration to: 100
16:19 Executing Touch 2 of 3 [0/10]
16:19 Touch 2 result: 406.0490
16:19 Reset Acceleration to: 10000
16:19 Deviation: 0.0062
      New Average: 406.0427
      Tolerance: 0.0100
16:19 Set Acceleration to: 100
16:19 Executing Touch 3 of 3 [0/10]
16:19 Touch 3 result: 406.0521
16:19 Reset Acceleration to: 10000
16:19 Deviation: 0.0094
      New Average: 406.0459
      Tolerance: 0.0100
16:19 Completed 3 touches with a standard deviation of 0.0068
16:19 Position Difference: 8.5469
      Adjusted Difference: 10.5531
16:19 Updated Initial Position: ['0.00', '0.00', '2.01', '0.00']
16:19 Offsetting by 0.050
16:19 Touch procedure successful with 0 retries.
16:19 Final position: [0.0, 0.0, 1.9562499999709757, 0.0]
16:19 Standard Deviation: 0.0068
```

krautech, 2024-12-06:
So the point is you need to set an offset now to which youre happy with the appropriate amount of first layer squish. TOUCH sets the Z0 point as ON your bed. Then we automatically have an offset of default 0.05, but to figure out whats perfect for you, during your first print you adjust z offset and save it.

The nozzle nor cartographer are unable to tell you what offset you need to get perfect first layer, it just gets a perfect Z0

Sab-tech-lab, 2024-12-09:
I added these two lines to the cartographer configuration to see if changing bed and running CARTOGRAPHER_TOUCH again would get a z-offset consistent with my print settings
```
cal_nozzle_z: 0.2
scanner_touch_z_offset: 0.18
```

krautech, 2024-12-09:
By changing the cal nozzle z you will effectively be putting calibration out of sync.

This is do at your own risk and we cannot guarantee you won't damage anything.

Sab-tech-lab, 2024-12-09:
So the two parameters that of used are deprecated?

krautech, 2024-12-09:
No, they're still there.
Cal nozzle z is just "adjust at your own risk"

And the other you should be adjusting via your UI not manually editing it

krautech, 2024-12-17:
@Sab-tech-lab hows this going?

Sab-tech-lab, 2024-12-17:
All works perfect for now.
After calibrating the machine, I set the z-offset as described in the documentation and have been printing for some time without any problems.
I am trying to create a macro to be run when changing the bed so that the set z-offset remains unchanged, I think it is enough to run the CARTOGRAPHER_TOUCH command but I haven't had time to try it yet. 
This is the repo with my configuration:
https://github.com/Sab-tech-lab/Cartographer3d-FLsun-V400