Cartographer3D/cartographer-klipper #161: Fix non critical mcu issue with starting klipper where the cartographer is not connected.

pellcorp, 2024-12-08:
Currently if you configure the scanner to be `is_non_critical: True`, and do not have it connected at startup, you get this very confusing error message:

[image]

The reason this occurs is because at line 1224:

```
self.inv_adc_max = 1.0 / constants.get("ADC_MAX")
```

The constants map has no value because the scanner has not been connected and because of is_non_critical this is ok, but the scanner.py does not cater for it.

Jomik, 2024-12-08:
On `develop` this should already be improved. You'd get the, still technical, "Firmware constant 'ADC_MAX' not found.
We have improved that slightly in #170.
It should show "Error during probe mcu identification, check connection: Firmware constant 'ADC_MAX' not found"n

Please let us know what you think.

pellcorp, 2024-12-08:
That error is going to confuse the hell out of Simple AF users.   A 'scanner is not connected' error is much easier to understand, but I also understand its not a priority to support danger klipper users who have this field on and their carto misconfigured.

pellcorp, 2024-12-08:
I can I guess add docs around it at least

krautech, 2024-12-08:
Yeah the issue comes from when that error triggers, it can be caused by alot of things and we need to be able to pinpoint that error to be able to fix it.

Annoyingly, as I can see the appeal. 

Might be a better way to look for scanner is disconnected but might be more involved (actually looking for the USB device to see if it's missing etc) 🤔

pellcorp, 2024-12-09:
I actually think this will work fine, I have updated the docs, I am going to close this issue, I think its fine as is, my docs are

[image]

closed 2024-12-09 by pellcorp