Cartographer3D/cartographer-klipper #168: Klipper error after update

ModularPrintingSystem, 2024-12-08:
Since I updated klipper I get following error:
[image]

krautech, 2024-12-08:
Did you check troubleshooting at all? 🤔

https://docs.cartographer3d.com/cartographer-probe/troubleshooting

ModularPrintingSystem, 2024-12-08:
Thank you, I didn't look at it, but after following the troubleshootinh guide I got another error, that I couldn't fin
[image]
d

krautech, 2024-12-08:
Lis2dw probes are currently broken thanks to klipper.

You need to comment out the [lis2dw] section

This is fixed in 1.1.0 which will be released in the next 24-ish hours.

ModularPrintingSystem, 2024-12-08:
Okay, thank you

closed 2024-12-08 by ModularPrintingSystem