Cartographer3D/cartographer-klipper #177: CARTOGRAPHER_TOUCH disapear

hougoul, 2024-12-08:
After upgrade of Klipper and Cartographer to the last version.
After applied the probe register fix
I cannot do any command CARTOGRAPHER_TOUCH anymore.

Jomik, 2024-12-08:
Please help us help you.
How are we supposed to do anything when you provide us absolutely no information.
`CARTOGRAPHER_TOUCH` works fine here.
We will need your `klippy.log`, preferably roll it before you upload it so that it is small.
Issues like this are easier to resolve via [Discord](https://discord.com/channels/1165274913624572014/1229798364514750596).

hougoul, 2024-12-08:
Sorry,
I will try to give it as soon as possible. But I think it's because scanner.py is not in my klipper/klippy/extra folder.

krautech, 2024-12-08:
Ahh you need to do

`cd ~/cartographer-klipper
./install.sh`

liangw1024, 2024-12-13:
I ran CARTOGRAPHER-TOUCH, and after 10/10, it prompted that the nozzle had not reached the extrusion temperature
My touch spray temperature is 145 ° C

Jomik, 2024-12-13:
@liangw1024 Please create your own issue, as it is not the same problem. Or preferably, open a thread in our Discord.
https://discord.com/invite/EFSwpureHz
https://discord.com/channels/1165274913624572014/1229798364514750596

closed 2024-12-17 by krautech