Cartographer3D/cartographer-klipper #205: [flasher] dont default to full firmware

yell3D, 2024-12-16:
Flash by default only Klipper not also Katapult. When flashing full firmware if something goes wrong you need to resort to dfu flash.

Propose:
* only offer full when there is a hard requirement to do so
* default to survey klipper firmware
* add option to advance to ignore this and freely choose to flash full, katapult, klipper

krautech, 2024-12-17:
Only DFU flashes full firmware? The others do not.

I made DFU only flash full firmware and not either klipper, katapult or both because there isnt really a necessity for flashing just klipper. 

The added benefit is it will fix any broken katapult installations by default rather than having people try diagnose.

As well as it would be an extra step for people who need to use DFU because for some reason katapult isnt working.

I cant see a use case where flashing just klipper via dfu would be necessary.

Happy to be proven otherwise however.

Esoterical, 2024-12-17:
yeah, @yell3D could you offer an elaboration on what you're wanting fixed?
As it stands, flashing via katapult only writes the klipper.bin to the carto, doesn't touch katapult itself.
Flashing via DFU, why not just do a full katapult + klipper flash. 
There's no downside (that I can see) of doing a a combined firmware flash if you're already in DFU mode, and I can't think of a scenario where this would cause issues.

yell3D, 2024-12-17:
So the `[ull_Survey_Cartographer_CAN_1M_5_0_0.bin](https://github.com/Cartographer3D/cartographer-klipper/blob/master/firmware/v2-v3/combined-firmware/5.0.0/Full_Survey_Cartographer_CAN_1M_5_0_0.bin)` isnt a Full firmware when its flashes via Katapult? 

Because thats what the flasher told me it  flash and afterward my HT didnt react to klipper (wrong firmware i know) but also didnt reset to katapult mode so i assumed it flashed both.

yell3D, 2024-12-17:
#WontFix #UserError #NoSafetyWheels

closed 2024-12-17 by yell3D