Cartographer3D/cartographer-klipper #209: Getting Ècoil expected max frequency exceededÈ

andremalenfant, 2024-12-17:


morlan, 2025-02-11:
I'm also intermittently getting this error which causes Klipper to shut down.  I heat soak my printer and it crashes during homing or QGL.  Sometimes also occurs after an Emergency Stop and reboot.

real-artswan, 2025-02-28:
It seems it happens each time when klipper tries to use the probe and toolhead MCU (I have EBB SB2209 CAN V1.0（RP2040)) temperature is more than 80C. I made kind of a workaround - a macro before homing Z that moves head close to the aux fan and turn on it together with a part fan on 100% and wait while the MCU temp < 80. 
Don't know is it cartographer issue.

real-artswan, 2025-03-01:
Still get the error even if temps are low (around 40C). It never happens on the very first print after powering on, but on some of the next prints. When it happens "FIRMWARE_RESTART" just does nothing, I have to switch power to recover it.

editwentyone, 2025-03-04:
i am in the same boat now. same behavior, print died mid way and now homing brings me this message

dancollins, 2025-03-08:
> Still get the error even if temps are low (around 40C). It never happens on the very first print after powering on (even on temps higher than 80), but on some of the next prints. When it happens "FIRMWARE_RESTART" just does nothing, I have to switch power to recover it.

I am experiencing the same. After boot, it doesn't seem to be an issue. After a print or, at the moment, after a `RESTART` is enough to reliably trigger it.

I'll keep an eye on the carto (and coil) temperature to see if there's a pattern there.

dancollins, 2025-03-09:
Today I've been playing with the Klippain ShakeTune tool, and that has had me get the printer hot but with the hotend fan turned off. In this configuration, the toolhead is slightly cooler (just below 80 °C, instead of at or above 80 °C). Reproducing using the steps above has had a high success rate - but rebooting (full power cycle) is enough to recover even when the temperature is the same.

real-artswan, 2025-03-10:
My hardware:

```
mcu(stm32h723xx)
Version: v0.12.0-451-g730e5951b

mcu EBBCan(rp2040)
Version: v0.12.0-451-g730e5951b

mcu scanner(stm32f042x6)
Version: CARTOGRAPHER 5.1.0

Host(aarch64, 64bit)
Version: v0.12.0-456-gfbd5b492-dirty
OS: Armbian 25.2.3 bullseye
```

every mcu seems updated and synced

real-artswan, 2025-03-11:
Unfortunately no, hot coil and mcu still makes an error

real-artswan, 2025-03-19:
again and again it happens. I think the problem is in design, don't know if it can be compensated in software. When MCU on the carto is hanging/flying over the hot bed it's started to glitch when heating.It might not happen on the very first print but will happen on second. Another thing that "FIRMWARE_RESTART" doesn't work when it happens. You have to switch power.

real-artswan, 2025-03-19:
Sometimes I'm gettinh instead "Error during homing probe: Communication timeout during homing"

real-artswan, 2025-03-19:
again and again it could be suitable for PLA/PETG, but doesnt work for ABS when bed is 100 and chamber 45 and carto MCU has 75+. Wrong design. MCU must not be over the hotbed if it's so sensetive.

dancollins, 2025-04-06:
> again and again it could be suitable for PLA/PETG, but doesnt work for ABS when bed is 100 and chamber 45 and carto MCU has 75+. Wrong design. MCU must not be over the hotbed if it's so sensetive for temp, it's obvious things, right?

I think you are mistaken here. The microcontroller, an STM32F046x6, is rated to run with a junction temperature of 105 °C. Mine is several hours into an ABS print right now, and is sitting at 90 °C which is about as high as I've seen it. These maximum values often have some margin, as it's the minimum temperature ST will gaurantee their parts will operate at (i.e. many will likely perform higher than this).

WRT the other components, the LDC1604 and the CAN transceiver, I would assume they are rated similarly. Feel free to check the datasheets if you want further confident.

I suspect the coil measurement is failing for some reason, and then recovering from that fault is where the issue lies. Talking to Richard on Discord, the part seems to go outside of its datasheet parameters and becomes unresponsive on the I2C bus. Unforunately, Klipper has no choice but to crash in that scenario - as the sensor can't be recovered. Whether that's hardware or firmware is hard to say.

real-artswan, 2025-04-13:
Thanks for the response.
Yes, datasheets must allow these temperatures, and even more, it happens even if I cooldown everything. It just happens each second or third print and then you have to switch off the power to start again. 
But I replaced it with eddy coil and have no any issues, so not sure if I can help further.

morlan, 2025-04-13:
I just converted from USB to CANBUS and getting these errors every time I want to home with the heaters or cameras on. I’m still trying to isolate which one is the culprit.

I actually don’t think it’s due to temperature rating of the device because my chamber and everything was already heat soaked. It’s just when actually having the heaters or cameras are running. So I believe it’s something with the bandwidth or load. 

Temporary workaround for me is just to heat soak, turn off heaters and/or crowsnest, then home.

skyguy94, 2025-04-27:
I flashed my cartographer from FW5.0.0 to FW5.1.0 to see if it would help with my occasional nozzle touch offset failures. After flashing, when attempting to recalibrate, I consistently get this error: 

`Scanner hardware issue: coil expected max frequency exceeded`. 
```
9:51 AM Klipper state: Shutdown
9:51 AM Scanner calibration starting
9:51 AM ACCEPT
9:51 AM Z position: ?????? --> 207.000 <-- 208.000
9:51 AM TESTZ Z=-1
```

I cannot recover from this error by using the restart commands only by restarting the klipper service.

My cartographer is connected via CAN to a EBB SB2209 (RP2040) board and I have ~1500 hours of printing ABS in a heat soaked chamber with cartographer as my probe. I was heatsoaking while attempting to calibrate initially, but given the previous post, I turned off and let the system cool, but I still get the message every time I try to calibrate with FW5.1.0.

Rolling back to firmware 5.0.0 resolved the problem. For those who aren't aware, to rollback:
* Run `./scripts/firmware.py` from the cartographer-klipper folder on your system.
* Select (4) Enable Advanced Mode
* Select (13) Enable Show All Firmware
* Select the appropriate flashing mode (1 or 2/CAN or USB)
* Select (1) to update your cartographer
* Select (2) the firmware. You should see 5.0.0 and 5.1.0 in the list after the tarball has downloaded.
* Select the 5.0.0 firmware (for me this was option 1)
* Select (4) Flash Selected Firmware

morlan, 2025-05-02:
Thanks for the instructions to downgrade the firmware to FW5.0.0.  It seems to have resolved the problem for me as well.

drewgwallace, 2025-05-27:
> Rolling back to firmware 5.0.0 resolved the problem. For those who aren't aware, to rollback:
>
>     * Run `./scripts/firmware.py` from the cartographer-klipper folder on your system.
>
>     * Select [...]

@skyguy94  thank you! You've solved a massive headache!

From my observations the "Scanner hardware issue: coil expected max frequency exceeded" issue is correlated to a heated chamber, reset by a full power cycle (I turn off mains power ~10 seconds). A reboot isn't enough. A downgrade to v5.0.0 was the best option until the problem is acknowledged and fixed.