Cartographer3D/cartographer-klipper #223: Cartographer_touch gives internal error when using bed_tilt

harshitgoel96, 2025-01-20:
Hi,

I am trying to use Cartographer as a Z end stop and for auto Z. I want to use bed_tilt_adjust with cartographer_touch. My concern here is solely for auto z feature of cartographer.

When i define a bed_tilt config, and run cartographer_touch I get an error in klippylog as follows

```
Internal error on command:"PRINT_START"
virtual_sdcard dispatch
Traceback (most recent call last):
  File "/home/printer/klipper/klippy/extras/virtual_sdcard.py", line 273, in work_handler
    self.gcode.run_script(line)
  File "/home/printer/klipper/klippy/gcode.py", line 230, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/printer/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/printer/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/printer/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/printer/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/printer/klipper/klippy/gcode.py", line 227, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/printer/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/printer/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/printer/klipper/klippy/extras/scanner.py", line 456, in cmd_SCANNER_TOUCH
    vars = self._get_common_variables(gcmd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/printer/klipper/klippy/extras/scanner.py", line 441, in _get_common_variables
    "TOUCH_LOCATION_X", float(self.touch_location[0])
```

I think this might be due to zero_reference_position not available under bed_tilt config, as its hard tied to bed_mesh.

I think if me move ZRP to scanner config, cartographer_touch can run without a bed_mesh. 
Is there a way to provide ZRP outside of bed_mesh?

krautech, 2025-01-20:
Interesting, as a work around can you just create the following?

```
[bed_mesh]
zero_reference_point: x, y (change these)
```

Or does it require you to have all the options filled out?

harshitgoel96, 2025-01-20:
[bed_mesh] and [bed_tilt] both can not exist together. 

from Klipper docs
```
[bed_tilt][¶](https://www.klipper3d.org/Config_Reference.html#bed_tilt)
Bed tilt compensation. One may define a bed_tilt config section to enable move transformations that account for a tilted bed. Note that bed_mesh and bed_tilt are incompatible; both cannot be defined.
```
Ref : https://www.klipper3d.org/Config_Reference.html#bed_tilt