Cartographer3D/cartographer-klipper #235: Touch mode

madnezzm, 2025-03-05:
How make bed leveling and bed mesh in touch mode?

Touch mode only working on z offset.

In scanning mode on pei sheet the results are incorrect, It shows bumps in the same place, but they are not there.. Checked in manual mode paper test.

szc188, 2025-05-20:
Hey, I’ve been dealing with the same issue! 

After swapping out a bunch of different PEI sheets from various brands, it got slightly better, but the scanning errors are still around 0.1mm. 

With Klicky, though, it’s way tighter at 0.03mm. 

During the first layer, some spots have gaps while the center sticks way too hard. 

Right now, I’m just printing with BED_MESH turned off—it helps a little but isn’t perfect. 

The touch mode is super accurate, even if it’s slower. Honestly, I’m okay with the speed trade-off for now to temporarily deal with these uneven PEI sheets. Just gotta live with it until I find a real fix.

This is my current bed mesh:

[image]

pixel-shock, 2025-08-13:
+1 for me ... i have exactly the same issue with the scan mode ... currently my printer isn't usable at this state ... I am not able to get a proper first layer (1st time in years) - coming vom CNC Tap

dobromyslov, 2026-08-24:
Here are my results for `BED_MESH_CALIBRATE PROBE_METHOD=touch`.

### 1. Touch 15x15 without interpolation (GOOD)
- range 0.176mm
- time 17 minutes
[image]

### 2. Touch 9x9 with interpolation `mesh_pps: 2,2` (BAD)
- range: 0.167mm
- time: ?
- problem: I started it within 5 minutes after switching the heat-bed on, did not let the printer to fully heat-soak
[image]

### 3. Touch 7x7  with interpolation `mesh_pps: 2,2` (PERFECT)
- range 0.139mm
- 8 minutes
[image]

### Conclusion
7x7 touch survey with `mesh_pps: 2,2` interpolation perfectly correlates with the reference 15x15 mesh, smooths unnecessary noise and insignificant height deviations. Moreover, it takes only 8 minutes.

Here is the comparison of the first layers with different height-map approaches for a PEI plate with inconsistent magnetic profile:
- Left - `BED_MESH_CALIBRATE PROBE_METHOD=touch` (https://github.com/Cartographer3D/cartographer3d-plugin/pull/411). (there are two small spots caused by some greasy fingerprints on the plate)
- Right - `BED_MESH_CALIBRATE PROBE_METHOD=scan`.

[image]

dobromyslov, 2026-09-19:
By the way, I've solved the Y-axis horse saddle/taco problem by installing [titanium backers on the Y extrusions](https://github.com/VoronDesign/VoronUsers/tree/main/printer_mods/whoppingpochard/extrusion_backers?utm_source=gemini). Even though I installed a titanium backer plate on the X extrusion, the X deformation still persists.

[image]

I am happy that the Cartographer [bed mesh touch feature](https://github.com/Cartographer3D/cartographer3d-plugin/pull/411?utm_source=gemini) has been working perfectly for me all month. I hope it will be merged into the main branch soon.